xioTechnologies/Gait-Tracking #2: No imufusion import

dovanan95, 2022-03-02:
The imufusion library in gait_tracking.py is not available. Could you help to fix tt? Thanks so much!

xioTechnologies, 2022-03-02:
You need to first install the python package using `pip install imufusion`.

dovanan95, 2022-03-02:
I already executed command: pip install imufusion. However, I received the response:
"ERROR: Could not find a version that satisfies the requirement imufusion (from versions: none)
ERROR: No matching distribution found for imufusion"

dovanan95, 2022-03-02:
I finished installing imufusion lib, but I have to change to Windows OS. Thank you!

closed 2022-03-02 by dovanan95

grginho, 2022-03-18:
Hello,

Can you please build imufusion package for Linux?
If not, are you able to provide some instructions on how can it be done so we can do it?

xioTechnologies, 2022-03-18:
A Linux build is available.  You can see the available builds on this page:
https://pypi.org/project/imufusion/0.0.50/#files

grginho, 2022-03-18:
> A Linux build is available. You can see the available builds on this page: https://pypi.org/project/imufusion/0.0.50/#files

Thank you very much for such an urgent response. I was searching for it yesterday and I haven't checked if it is available today.

Best regards