xioTechnologies/Gait-Tracking #3: Drop on the Z-axis

grginho, 2022-04-01:
Hello @xioTechnologies ,

First of all, thank you for your great work on this Gait-Tracking algorithm.
I'm working on an IMU project using the STM32 lsm6dsr sensor, and I'm having issues with dropping over 1 meter on the Z axis (please take a look at the pictures on the links below).
[X-Y Plane.png](https://drive.google.com/file/d/1BcVGb12eEzXqVZt9mciKlYUAmYGko6ZI/view?usp=sharing)
[X-Y-Z Plane.png](https://drive.google.com/file/d/1pCV_wz7p7rjALFow7K3htr5o1Umaa2JU/view?usp=sharing)
Do you have an idea what can be issuing this dropping on the Z-axis, or what I can do to decrease it? The test was made on a flat surface, and it shouldn't be any movement around the Z-axis. I have used data from a gyroscope and accelerometer.

Thank you in advance and best regards,

Dragan

xioTechnologies, 2022-04-04:
I suspect that the vertical drift is due to inaccurate (uncalibrated) sensor measurements.  The algorithm removes the 1 g of gravity from accelerometer measurements to calculate a velocity and then position.  Inaccurate sensor measurements will mean that some residual of this 1 g remains in the measurements.  This residual vertical acceleration is the likely cause of the drift in your vertical axis.

grginho, 2022-04-06:
I've tried with the accelerometer calibration but didn't get much better results.
As per my knowledge, 1 g is removed in the Linear Acceleration data. So, I've plotted Linear Acceleration Data vs Acceleration data. Linear Acceleration data is collected from your algorithm using ahrs.linear_acceleration.
[image]
Is there any other way to confirm if the issue is caused by a residual of 1 g on accelerometer data?

xioTechnologies, 2022-04-07:
Your plots show a vertical drift of approximately 100 mm per step, with the swing phase of each step lasting approximately 1 s.  This corresponds to an acceleration offset error of 10 mg.  I expect you will see this offset in the acceleration plot generated by [line 92](https://github.com/xioTechnologies/Gait-Tracking/blob/main/gait_tracking.py#L96).

Consider that a 10 mg acceleration offset error in the vertical axis is equivalent to an orientation measurement error of 8 degrees.  Both accelerometer and gyroscope measurement errors may be contributing to the error in acceleration.

grginho, 2022-04-07:
I tried to visualize that offset of 10 mg, using plot from line 96, and just removing Earth's gravity constant to get acceleration in g instead of m/s/s. I've compared data from your dataset short_walk.csv and my own data user_data.csv zooming the stationary part of one step. 
Values of the accelerometer data on all three axes are not constant during this stationary time (they change in a range of 0.02:-0.02). and it is hard to tell if there is an offset on any of them. This is the case in both data sets. 
[image]
Both data seem to be really similar and I didn't manage to see that offset. Is this the right way to visualize it?
Maybe, this problem is caused because of the gyroscope, which is the best way to check if that is the case?

xioTechnologies, 2022-04-07:
You have created the visualisation I suggested and your comparison with **short_walk.csv** is useful.  The source of the vertical drift appears to be more complicated than an offset error in the acceleration.

Notice the large acceleration spikes in **user_data.csv** that are not present in **short_walk.csv**.  The start of each stride in **short_walk.csv** is a smooth curve as the foot rocks forward, shifting the body weight onto to toes.  The start of each stride in **user_data.csv** is very different, with spikes that vary significantly between strides.

I suggest you improve your IMU mounting on the foot to avoid these spikes and to achieve a consistent acceleration profile between strides.  **short_walk.csv** was collected with the IMU bound tightly to the laces of the shoe.  A loosely mounted IMU will bounce and oscillate during walking, introducing high-frequency content that will ultimately be integrated as drift.

grginho, 2022-04-12:
I’ve improved results by placing a sensor under my shoelaces. I have also run calibration algorithms for accelerometer and gyroscope and got almost identical results as you did.
But my use case is a bit specific and I need to wear a sensor on the shin (the outside of the lower leg). Placing the sensor in that way, I have a drop close to 1m on a Z-axis with the same number of steps.
Is there any suggestion or a bit of advice on what I can do to improve results with this sensor placement?

Thank you in advance

xioTechnologies, 2022-04-13:
I am pleased that you can now achieve almost identical results.  Modifying the algorithm for an IMU mounted on the leg instead of the foot is beyond the scope of support I can offer here.

IKetchup, 2022-04-15:
Hello,
First I would like to thank you for the amazing work you did with this library.

I am having the same problem of Z-axis drop. I am mesuring acceleration and angular speed with a smartphone with "physics tools box suite" (sensors calibrated). I have tried different scenario where I walk with the phone in my hand but also where I move the phone in circle on a flat surface.

In both case I have a terrible drop in Z-axis. Here is the exemple where I move the phone on a flat surface (circle of around 30cm) : 
[image]

[image]

Here is the exemple where I walk with the phone in my hand (walk in circle around 25 step) : 

[image]

[image]

In the second scenario I also have a problem of dimension : the position move less than a meter but in reality the phone have moved from around 5-6m.

Do you have any ideas where the drop in Z-axis and the problem of position come from? My final goal would be to visualize the 3D trajectory of an object falling from the sky.

Thanks in advance,

xioTechnologies, 2022-04-15:
The algorithm is for an IMU mounted on the foot.  It will not work for an IMU held in the hand or for an object falling from the sky.

closed 2022-04-15 by xioTechnologies