xioTechnologies/Gait-Tracking #8: Does the gain-tracking algorithm work even if exercising at constant speed?

Taeyoung96, 2022-10-10:
@xioTechnologies   
Hi, Thanks for your great work.  
I'm going to do a dead reckoning using IMU in a mobile robot.  

For this algorithm, I understand that speed and position are obtained through integration only if there is acceleration.  
For constant velocity motion, the acceleration value is very small.  

Then can I use this algorithm even in the above situation?  
If it's not suitable, do you have any algorithms to recommend in my situation?  

I'd appreciate it if you could answer me.  
Thanks,

xioTechnologies, 2022-10-10:
You are unlikely to achieve useful measurements of position using an IMU alone.  Consider that even just a 1 degree inclination error in the orientation measurement would result in an acceleration measurement error of 1g * sin(1) = 0.171 m/s/s.  This would result in a position error of 8.5 meters after 10 seconds of dead reckoning.  IMUs are typically only used for positional measurement when combined with other technologies such as GNSS.

Taeyoung96, 2022-10-11:
@xioTechnologies  
Thank you for your quick and kind reply.  
I will find a way to apply other sensor fusion methods.  

Thanks,

closed 2022-10-11 by Taeyoung96