xioTechnologies/Gait-Tracking #14: Module 'imufusion' has no attribute 'CONVENTION_NWU'

Alx2203, 2023-04-28:
Hello, thank you very much for your project which helps me a lot. However I wanted to test your code and I get this error:

AttributeError: module 'imufusion' has no attribute 'CONVENTION_NWU'

I'm sorry if this is a silly question but I'm a beginner. I checked that the latest version of imufusion was installed. I'm working on conda, maybe that's where the problem comes from?

Thanks in advance for your help.

xioTechnologies, 2023-04-28:
I expect you are not using the latest version of [imufusion](https://www.piwheels.org/project/imufusion/).  I am not familiar with conda so I am not able to suggest a solution.  I suggest you search online.  There many Python guides and tutorials for beginners.

Alx2203, 2023-04-29:
Ok I solved my first problem. Now I have a second one x)
I tested the code with your short_walk and long_walk data and it worked perfectly. When I test with my IMU I consistently get an increase along the z axis while my data was obtained flat.
[image]
[image]

Other than a calibration problem, do you have any idea where the problem is coming from?

xioTechnologies, 2023-04-29:
Other users have asked similar questions.  Please read through the closed issues for this repository.

Alx2203, 2023-05-01:
Sorry I didn't pay attention to the closed issues.
Ok it also seems that my problem was also due to the IMU being too loose yet I was already attaching the IMU to the laces.
By the way, where exactly do you place the IMU at the laces (rather close to the toes or the ankle)? Do you have any tips on how to make sure the IMU is tightly attached?

xioTechnologies, 2023-06-14:
Above the toes or on the laces of a shoe are often convenient locations.  I suggest using an elasticated strap or if the show is big enough then tucking the IMU between the tongue and the laces.

closed 2023-09-12 by xioTechnologies