xioTechnologies/Gait-Tracking #16: some issues with imufusion settings and other questions

pier098, 2023-10-09:
Hello,

I'm using your code to plot a 3D figure of gait tracked with my Arduino MKR IMU shield (BNO055), but I'm encountering some problems. In particular, I believe that the plotted position (in meters) does not match reality.

First of all, I collected 10Hz data for over 60 minutes without filtering the data in my Arduino code. Do I need to filter the data?

The IMU was upside-down, so the z-axis was pointing downwards. I set the NED convention, but do I need to multiply the acceleration by -1?

These are my imufusion settings:
`
``ahrs.settings = imufusion.Settings(imufusion.CONVENTION_NED, 0.8, 2000, 20, 10, 1 * sample_rate)```

And I modified the acceleration threshold as follows:

`is_moving[i] = numpy.sqrt(acceleration[i].dot(acceleration[i])) > 0.55 

and the margin :

`margin = int(0.1 * sample_rate)`

Could you please tell me if I have set the parameters correctly?

Thank you.

xioTechnologies, 2023-10-09:
"_Do I need to filter the data?_"
No

"_do I need to multiply the acceleration by -1_"
No

"_Could you please tell me if I have set the parameters correctly?_"
The original parameters already work well for the example data and should work for any data of sufficient quality to be used in this application.

I suggest you increase your sample rate to 100 Hz and start by logging just a few steps in a straight line.  You may also find it useful to look through closed issues to see what other people have shared when using this code.

xioTechnologies, 2024-02-22:
I expect you will get very poor results with a sample rate of 10 Hz.

susovanpatra00, 2024-03-09:
> Thank you so much for your answer !
> 
> Does your filter work well for my purpose if I only use the IMU (i.e. 3D visualizing of position for a long time) ? I'm asking because in your PhD thesis, it's written that it could be used with cheap sensors and it works well. However, I've read a closed issue where you advised integrating it with GNSS

His man,,, is it working for you??

closed 2024-08-27 by xioTechnologies