xioTechnologies/Gait-Tracking #18: Error: 0.082 m

Rocky-YU, 2023-11-07:
Hello,
Thanks for sharing your code, I am getting some errors while running your code. For the parameters and data of the code I did not make any changes. But still this error occurs.
"
D:\ProgramData\Anaconda3\envs\labelme\python.exe D:/code/Gait-Tracking/gait_tracking.py
Error: 0.082 m
"
What should I do about it, please?
Thank you.

xioTechnologies, 2023-11-07:
That is not a runtime error.  It is a printed message indicating the calculated error in the final position in meters.

Rocky-YU, 2023-11-07:
I see, thank you.

Rocky-YU, 2023-11-07:
> That is not a runtime error. It is a printed message indicating the calculated error in the final position in meters.

I see, thank you.

closed 2023-11-07 by Rocky-YU