xioTechnologies/Gait-Tracking #19: Can this algorithem used in Android's IMU data

foreee, 2023-11-29:
Hi , can this tracking method used in the Android data? 

I tryed to get the gyroscope and accelerometer data from Android phone, some data like below
```
Time (s),Accelerometer X (m/s/s),Accelerometer Y (m/s/s),Accelerometer Z (m/s/s),Gyroscope X (rad/s),Gyroscope Y (rad/s),,Gyroscope Z (rad/s)
0.005 0.16660027 0.07058827 9.853045 -0.0017561859 -0.0018325418 3.8177954E-4 
0.01 0.1689931 0.07537391 9.83151 -5.344914E-4 -0.0012216945 -2.2906772E-4 
0.015 0.17377874 0.0634098 9.836296 -0.0017561859 0.0012216945 -0.0014507623 
0.02 0.14745769 0.058624156 9.8219385 -0.0023670332 0.0 -8.39915E-4 
0.025 0.14027923 0.04426722 9.836296 -5.344914E-4 -0.0018325418 -0.0014507623 
0.030000000000000002 0.14985052 0.032303106 9.81476 -0.0023670332 -0.0012216945 -8.39915E-4 
0.035 0.15702899 0.032303106 9.843474 7.635591E-5 -0.002443389 -8.39915E-4 
0.04 0.15463616 0.051445685 9.802796 -5.344914E-4 -0.0012216945 -2.2906772E-4 
0.045 0.1689931 0.05383851 9.829117 7.635591E-5 -0.0018325418 -2.2906772E-4 
```

the sample_rate is 200HZ and I changed the data fomat  by follow code
```
sample_rate = 200  # 200 Hz

timestamp = data[:, 0]
gyroscope = data[:, 4:7]*180/math.pi
accelerometer = data[:, 1:4]/9.81
```

I first stood still for a period of time, then walked around a building with my phone, which was about tens of meters. In the end, the starting and ending positions were basically the same. However, from the calculation results, there was basically no change in the displacement.
[image]

xioTechnologies, 2023-11-29:
Your CSV headings do not match the example data.  You have re-ordered the columns and used **m/s/s** instead of **g**.

foreee, 2023-11-30:
> Your CSV headings do not match the example data. You have re-ordered the columns and used **m/s/s** instead of **g**.

Yes, I have preprocessed the data after read, as the follow code
```
sample_rate = 200  # 200 Hz

timestamp = data[:, 0]
gyroscope = data[:, 4:7]*180/math.pi
accelerometer = data[:, 1:4]/9.81
```

xioTechnologies, 2023-11-30:
I'm sorry, I did not see that modification.  I suggest you read through the closed issues in this repo and in [Gait-Tracking-With-x-IMU](https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU) and [Gait-Tracking-With-NGIMU](https://github.com/xioTechnologies/Gait-Tracking-With-NGIMU) to see how other users have successfully used the code.

closed 2024-02-22 by xioTechnologies