xioTechnologies/Gait-Tracking #21: a drop in Z-axis

zhang1796183796, 2024-02-26:
hi,i have met a problem ，When I tested using my own data, there was a deviation of g in the acceleration of the z-axis. How should I eliminate it。
[image]

zhang1796183796, 2024-02-26:
[image]

zhang1796183796, 2024-02-26:
[image]

xioTechnologies, 2024-02-26:
The issue is specific to your system but it is not possible to comment further without more information.  The code works with the example data provided.  The example data was collected using an [NGIMU](https://x-io.co.uk/ngimu/).

zhang1796183796, 2024-02-27:
> The issue is specific to your system but it is not possible to comment further without more information. The code works with the example data provided. The example data was collected using an [NGIMU](https://x-io.co.uk/ngimu/).

i just walk on the plane.but there is acceleration  drop aruond g on z axis.,in figure one .

xioTechnologies, 2024-02-27:
The accelerations very high.  Perhaps you are provided measurements in m/s/s when g is expected.

closed 2024-08-27 by xioTechnologies