xioTechnologies/Gait-Tracking #22: Direction Problem

susovanpatra00, 2024-04-03:
I have Acceleration data and Gyroscope data with 100Hz. When I am trying to use this the direction showing is total opposite than the actual direction. Why is that so?? 
And what are the parameters I should vary to make it compatible with my sensor(MPU6050 GY-521) ??

xioTechnologies, 2024-04-03:
An accelerometer and gyroscope alone cannot provide a measurement of magnetic heading.  The direction of heading will instead be arbitrary.

susovanpatra00, 2024-04-03:
Okk,,,so is there any way to find the appropriate direction also?? If I add magnetometer data in this code will it work then?? 
If you have something else in mind which might help me please tell me

xioTechnologies, 2024-04-03:
The inclusion of a magnetometer will allow allow measurements of heading to be relative to magnetic north.

susovanpatra00, 2024-04-03:
ok,,,but there is a case where when is_moving[index] value let's say 4 and for that I get a direction,,,but when I change that value let's say 4.5 the direction is getting totally opposite(180 degree)...why is that??

susovanpatra00, 2024-04-08:
I am sorry that I couldn't make it clear my point. Actually I was talking about this line - 
``` for index in range(len(timestamp)):
    is_moving[index] = numpy.sqrt(acceleration[index].dot(acceleration[index])) > 4.5 # threshold = 3 m/s/s
```
    
   
I need one more help. Let's say I have magnetometer data then what changes do I need to do in this section ??
```
delta_time = numpy.diff(timestamp, prepend=timestamp[0])

euler = numpy.empty((len(timestamp), 3))
internal_states = numpy.empty((len(timestamp), 3))
acceleration = numpy.empty((len(timestamp), 3))

for index in range(len(timestamp)):
    gyroscope[index] = offset.update(gyroscope[index])

    ahrs.update_no_magnetometer(gyroscope[index], accelerometer[index], delta_time[index])
    
    euler[index] = ahrs.quaternion.to_euler()

    ahrs_internal_states = ahrs.internal_states
    internal_states[index] = numpy.array([ahrs_internal_states.acceleration_error,
                                          ahrs_internal_states.accelerometer_ignored,
                                          ahrs_internal_states.acceleration_recovery_trigger])

    acceleration[index] = 9.81 * ahrs.earth_acceleration  # convert g to m/s/s
```

**Please Help !!**

xioTechnologies, 2024-04-08:
You would replace `ahrs.update_no_magnetometer` with `ahrs.update` as per the examples and documentation in [Fusion](https://github.com/xioTechnologies/Fusion).

closed 2024-08-27 by xioTechnologies