xioTechnologies/Gait-Tracking #23: other sensors

SimoneCaiati, 2024-08-21:
how can i use this code with data collected from a different sensor? i use a software called “Physics tool box” from my cell phone to collect the data.
or can this algorithm only work with data collected from an NGIMU sensor?
i report my two classes in the files. what am i doing wrong?
[file.zip](https://github.com/user-attachments/files/16691056/file.zip)

xioTechnologies, 2024-08-27:
The code will work with any sensor provided that the data format matches the CSV file, and the sensor quality/accuracy is sufficient.

SimoneCaiati, 2024-08-28:
I use “accelerometer-ism6dsm” for acceleration, the “gyroscope-ism6dsm” for gyroscope, and “akm-akm09918” for magnetometer. Are they sufficiently accurate as sensors?

SimoneCaiati, 2024-08-28:
it is wrong to use linear acceleration, right? because I am currently using this, but the results I am getting are not as expected.
Also I use “earth_acceleration” without converting to m/s^2, since the acceleration I register already is. is this correct ?

xioTechnologies, 2024-08-28:
The units of the measurements are indicated in the CSV headings: g and degrees/s.  There is no magnetometer.  It sounds like you are using an uncalibrated sensor and can therefore expect poor performance.

Liam00Lu, 2024-09-20:
Hi there. I reviewed the "short walk" and "long walk" files you provided. I noticed that the acceleration in all three axes during the stationary periods before and after the movement is approximately (0,0,1). After integrating the data, there is no displacement in the z-axis, which indicates that you only performed two-dimensional planar movement. So, I would like to ask if you have processed data involving three-dimensional movement? Additionally, at the initial moment, the sensor's coordinate system does not coincide with the Earth's coordinate system, and during the stationary periods, the sensor's acceleration values on all three axes are not zero due to the influence of gravity components. Please take a look at the data I uploaded. The sampling frequency is 100 Hz.

The method I can think of is to calculate the initial coordinate system of the sensor based on the three-axis acceleration data during the initial stationary period. Then, using the gyroscope data, rotate all the acceleration data into this reference coordinate system. If it's necessary to rotate into another reference coordinate system later on, a second rotation can be performed.

[Example.xlsx](https://github.com/user-attachments/files/17073419/Example.xlsx)

closed 2024-09-20 by xioTechnologies