xioTechnologies/Gait-Tracking #24: Coordinate System Rotation

Liam00Lu, 2024-09-21:
Hi there,
Thanks for sharing the algorithm and all your work!

After comparing your acceleration data with mine, I encountered some questions and uncertainties. I would greatly appreciate it if you could provide some insights and suggestions regarding the issues I've outlined below.

I reviewed the short walk and long walk data you provided, and the triaxial acceleration data during the stationary periods before and after the movement are mostly (0,0,1g). This suggests that the sensor's z-axis is almost aligned with the direction of gravity. Although there are changes in acceleration along the z-axis, the displacement after integration is quite small compared to the other two directions. This indicates that the object's motion is primarily two-dimensional.

So, I would like to ask if you have processed data involving three-dimensional movement? Additionally, at the initial moment, the sensor's coordinate system does not coincide with the Earth's coordinate system, and during the stationary periods, the sensor's acceleration values on all three axes are not zero due to the influence of gravity components. Please take a look at the data I uploaded. The sampling frequency is 100 Hz.

The method I can think of is to calculate the initial coordinate system of the sensor based on the three-axis acceleration data during the initial stationary period. In other words, we can calculate the angle between the sensor's coordinate system and the Earth's coordinate system based on the acceleration values when the sensor is stationary. Then, using the gyroscope data, rotate all the acceleration data into this reference coordinate system. If it's necessary to rotate into another reference coordinate system later on, a second rotation can be performed.

[Example.xlsx](https://github.com/user-attachments/files/17082711/Example.xlsx)

xioTechnologies, 2024-09-21:
I do not recognise you description of the accelerometer being [0, 0, 1] during the stationary periods.  Here is the accelerometer plot for [short_walk.csv](https://github.com/xioTechnologies/Gait-Tracking/blob/main/short_walk.csv).  The actual value is closer to [-0.5, 0.25, 0.8].

[image]

The motion in [short_walk.csv](https://github.com/xioTechnologies/Gait-Tracking/blob/main/short_walk.csv) and [long_walk.csv](https://github.com/xioTechnologies/Gait-Tracking/blob/main/long_walk.csv) is three-dimensional.  The walking in both cases is across a flat floor so the changes in height above the floor are significant less than the changes across the plane of the floor; but the motion is nevertheless three-dimensional.

Liam00Lu, 2024-09-22:
Thank you for your response and for pointing that out. I did misinterpret the accelerometer readings during the stationary periods, expecting them to be [0, 0, 1]. As you mentioned, the actual values are closer to [-0.5, 0.25, 0.8], as shown in your plot for short_walk.csv.

I had a couple of additional questions:

(1). Have you applied any high-pass or low-pass filtering to the accelerometer data? If so, how did you determine the cutoff frequencies?
(2). For more general motion data, such as the data I sent in the Excel file earlier, do you have any recommendations for an effective processing approach?

Thanks again for your clarification and assistance. I look forward to your feedback.

xioTechnologies, 2024-09-23:
The accelerometer data is not filtered.  The example is for tracking the motion of a foot during walking, it is not applicable to general motion.

Liam00Lu, 2024-09-23:
Got it. Thanks for your feedback!

chengbinwang, 2024-12-20:
> The accelerometer data is not filtered. The example is for tracking the motion of a foot during walking, it is not applicable to general motion.

I have same doubt,
In stationary periods, means not any move ,  why the data not like [0,0,1] , i got  [0,0,1],   is this because z-axis is not need aligned with the direction of gravity

closed 2025-02-13 by xioTechnologies