xioTechnologies/Gait-Tracking #25: data conversion

SimoneCaiati, 2025-01-20:
Hi! I'm trying to use the algorithm, and i noticed that for its use i've to use measurements that are in G and deg/s. Since I have m/s^2 and rad/s, is it okay if I convert them here and like this?

def process_sensor_data(self):
    self.delta_time = np.diff(self.timestamp, prepend=self.timestamp[0])
    self.euler = np.empty((len(self.timestamp), 3))
    self.internal_states = np.empty((len(self.timestamp), 3))
    self.acceleration = np.empty((len(self.timestamp), 3))

    for index in range(len(self.timestamp)):
        self.gyroscope[index] = self.gyroscope[index]* (180.0 / math.pi)            #rad/s in deg/s
        self.gyroscope[index] = self.offset.update(self.gyroscope[index])
        self.accelerometer[index]= self.accelerometer[index] / 9.81               #da togliere per short_walk

        self.ahrs.update_no_magnetometer(
            self.gyroscope[index], 
            self.accelerometer[index], 
            self.delta_time[index])

        self.euler[index] = self.ahrs.quaternion.to_euler()

        ahrs_internal_states = self.ahrs.internal_states
        self.internal_states[index] = np.array([
            ahrs_internal_states.acceleration_error,
            ahrs_internal_states.accelerometer_ignored,
            ahrs_internal_states.acceleration_recovery_trigger])

        self.acceleration[index] = 9.81 * self.ahrs.earth_acceleration  # convert g to m/s/s

Also, in one of the closed discussions, I read that this algorithm only works with IMU sensors attached to the foot, and not held in the hand. Did I understand this correctly or did I misunderstand?

xioTechnologies, 2025-01-21:
There is no issue in converting units.  The algorithm only works for an IMU attached foot.

SimoneCaiati, 2025-01-21:
is there a possibility of using it in such a way that it can also accept data from hand-held sensors?

SimoneCaiati, 2025-01-21:
all right, thanks so mutch

SimoneCaiati, 2025-01-21:
what if I found a way to strap the mobile phone to my foot firmly? because it seems that the algorithm works in such a way that it manages the periods of stasis and movement of the foot. If the data are accurate it should be fine, or not?

xioTechnologies, 2025-01-21:
I am not aware pf anyone testing this with a phone.  I expect it would be impractical but it should provide some success.  Phone IMUs will provide far less accuracy than the devices we use.

SimoneCaiati, 2025-02-13:
Hi! I'm testing the algorithm with data collected by mbientlab sensor placed on the foot during walking. I'm experiencing problems with the results obtained. My sensors are sampled at 100Hz, with acceleration in G and gyroscope in deg/s. why am i not getting a result that is faithful to what i expect? from the graphs it would seem that the measurements are taken correctly (i have also analysed them in excel).
I pass you one of the excel files with the results obtained; my movement is simply ‘a straight walk for about 3 metres’. I hope you can help me out, thanks in advance.

https://www.dropbox.com/scl/fo/zr4z5mj4of4y6q1eiaxt8/ALLzlIc7CH-HP_yqVkV-Tms?rlkey=qxbnble65qmcgaittg5vbcpa4&st=f2qs5j9g&dl=0

DWilliamsFF, 2025-02-25:
> This algorithm only works for an IMU attached to the foot.

What facets of the algorithm limit it to the foot?

DWilliamsFF, 2025-02-26:
> > What facets of the algorithm limit it to the foot?
> 
> The IMU must be stationary after each period of movement to reset the drift in the calculated velocity. The period of movement must be very short (e.g. ~1 second), and the stationary periods must be truly stationary. These conditions are typically only true for an IMU attached to the foot.

That makes a lot of sense. Thanks for clarifying